arduino/lede-yun #1: what i'm making wrong

scenaristeur, 2017-12-07:
not sure what to do to compile the image : 
(on Bash ubuntu on Windows 10) : 
i did this : 
david@DESKTOP-FGTO8A9:~$ ls
david@DESKTOP-FGTO8A9:~$ mkdir yun
david@DESKTOP-FGTO8A9:~$ cd yun/
david@DESKTOP-FGTO8A9:~/yun$ git clone https://github.com/arduino/lede-yun.git
Clonage dans 'lede-yun'...
remote: Counting objects: 402714, done.
remote: Total 402714 (delta 0), reused 0 (delta 0), pack-reused 402714
Réception d'objets: 100% (402714/402714), 133.11 MiB | 5.55 MiB/s, fait.
Résolution des deltas: 100% (276153/276153), fait.
Vérification de la connectivité... fait.
Extraction des fichiers: 100% (7120/7120), fait.
david@DESKTOP-FGTO8A9:~/yun$ cd lede-yun/
david@DESKTOP-FGTO8A9:~/yun/lede-yun$ make
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... failed.
Checking 'working-gcc'... failed.
Checking 'g++'... failed.
Checking 'working-g++'... failed.
Checking 'ncurses'... failed.
Checking 'zlib'... failed.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... failed.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python'... failed.
Checking 'git'... ok.
Checking 'file'... ok.
/home/david/yun/lede-yun/include/prereq-build.mk:157: recipe for target '/home/david/yun/lede-yun/staging_dir/host/bin/mkhash' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/david/yun/lede-yun/include/toplevel.mk:167: recipe for target 'staging_dir/host/.prereq-build' failed
make[2]: *** [staging_dir/host/.prereq-build] Error 1
/home/david/yun/lede-yun/include/toplevel.mk:83: recipe for target 'prepare-tmpinfo' failed
make[1]: *** [prepare-tmpinfo] Error 2
/home/david/yun/lede-yun/include/toplevel.mk:216 : la recette pour la cible « world » a échouée
make: *** [world] Erreur 2
david@DESKTOP-FGTO8A9:~/yun/lede-yun$

could you guide me . thxs

facchinm, 2017-12-11:
Hi @scenaristeur,
the prepackaged images can be found in the tftp folder; you should be able to flash it automatically using the link at the bottom of [this](http://forum.arduino.cc/index.php?topic=515869) post.

If you want to compile it yourself it looks like you are missing some dependencies (namely `gcc`, `python`  and `unzip`). The full list of dependencies needed by lede is [here](https://lede-project.org/docs/guide-developer/install-buildsystem) (the Debian9 instruction should apply, so you should run 
```
sudo apt-get install build-essential libncurses5-dev gawk git subversion libssl-dev gettext zlib1g-dev unzip
```

rmilecki, 2018-01-01:
> `/home/david/yun/lede-yun/include/prereq-build.mk:157: recipe for target '/home/david/yun/lede-yun/staging_dir/host/bin/mkhash' failed`

You've to start by looking at `prereq-build.mk:157`. You'll see that error happens for the following Makefile rule:
```
$(STAGING_DIR_HOST)/bin/mkhash: $(SCRIPT_DIR)/mkhash.c
	mkdir -p $(dir $@)
	$(CC) -O2 -I$(TOPDIR)/tools/include -o $@ $<
```
Either: `mkdir` or `$(CC)` command fails. You can try `make` debugging by executing something like `make -d` or a quick hack like modifying Makefile to:
```
$(STAGING_DIR_HOST)/bin/mkhash: $(SCRIPT_DIR)/mkhash.c
	mkdir -p $(dir $@)
	echo $(CC) -O2 -I$(TOPDIR)/tools/include -o $@ $<
	$(CC) -O2 -I$(TOPDIR)/tools/include -o $@ $<
```
You should see that Makefile is trying to execute command like
```
cc -O2 -I/home/david/yun/lede-yun/tools/include -o /home/david/yun/lede-yun/staging_dir/host/bin/mkhash /home/david/yun/lede-yun/scripts/mkhash.c
```
Error you'll see depends on what you're missing, **for me** it was failing with something like:
```
cc -O2 -I/home/david/yun/lede-yun/tools/include -o /home/david/yun/lede-yun/staging_dir/host/bin/mkhash /home/david/yun/lede-yun/scripts/mkhash.c
In file included from /home/david/yun/lede-yun/tools/include/endian.h:5:0,
                 from /home/david/yun/lede-yun/scripts/mkhash.c:82:
/home/david/yun/lede-yun/tools/include/byteswap.h:2:27: fatal error: byteswap.h: No such file or directory
compilation terminated.
```
You just have to see what the error is & fix it up.

facchinm, 2018-01-08:
It looks like you are missing `libc6-dev` from your target machine (or maybe `build-essential` which should contain most of the packages needed to build C programs)

closed 2018-11-23 by scenaristeur