arduino/lede-yun #3: Broken images on luci-webpanel

isantolin, 2017-12-15:
On LuCI lede-17.01 branch (git-17.315.61305-a9bddc4) / LEDEYun Reboot 17.11 r5461-ce9089c (Arduino-Webpanel) some images are broken, with src(unknown) in their img tags

(Image Attached)
[image]

@facchinm Can you fix this bug?

facchinm, 2017-12-18:
Hi @isantolin ,
indeed, the <img src> tag says unknown, but I can see the question marks...
[image]
Did you try using another browser? I'm on Chromium 62 but it should be ok in any browser.

isantolin, 2017-12-18:
@facchinm 
With any other browser works well. Just fails with Chrome 64.0.3282.24 (Versão oficial) beta 64 bits. (No console warnings)