arduino/lede-yun #4: difference between this config and OpenWRT18.06

DanielRIOT, 2018-08-16:
Hi, what are the differences between this repo and the one from Openwrt ( LEDE ) 18.06 in terms of the Yun's configuration, especially the serial console interface ?

I've built an image form 18.06 and it boots but  seems to reconfigure the baudrate at some stage during the kernel boot ( it starts at 250000 baud and one can see kernel messages then after about 2 seconds it goes to 9600 baud )

stty reports that 250000 is an invaid baudrate, but this can be set by manually running 

> reset-serial /dev/ttyATH0
 after boot ( from SSH ) or in rc.local.

How is the console configured in this version vs what is done in the mainstream Openwrt one so that the serial console works from this image , but not from the other one ?

-Ive also included the yun-config and yun-scripts in that build and built them into the kernel [*] and not as modules.

a forum post on is here, but no luck with Yun specific information from their side : https://forum.openwrt.org/t/arduino-yun-uart-console-baudrate-issue/19002/5
-Daniel

facchinm, 2018-08-20:
Hi @DanielRIOT ,
choosing 250000bps as baud rate was an unfortunate decision at the time but we have to live with that; there's a ugly patch that needs to be backported every time that enables that baudrate (https://github.com/arduino/lede-yun/blob/master/target/linux/ar71xx/patches-4.9/821-serial-core-add-support-for-boot-console-with-arbitr.patch) .
The reconfiguration to 9600 happens because bootargs request 250000 but fails (without that patch) and falls back to the minimum supported speed (9600).
I'd love to be able at some point to adopt mainline openwrt, so any advice or code could really help in this!

DanielRIOT, 2018-08-20:
Hi @facchinm 

that makes sense, 
a workaround I had is :
`serial-reset /dev/ttyATH0` in` rc.local` so it comes up later.

Thank you for pointing me to the patch, I updated it and did a test compile on 18.06 ( [this repo](https://github.com/openwrt/openwrt/tree/openwrt-18.06) ) and it seems to build happily- will test it when i get to a Yun tomorrow...

I also added these feeds to my feeds.conf so I can pull in yun-scripts[*] and yun-config[*] and yun-bridge[m].
`src-git arduino https://github.com/arduino/openwrt-arduino-packages.git;for-lede-next
`
after that one can follow the "regular" openwrt build process and all seems well ( I have gotten some "invalid image type" messages when doing sysupgrade, and will hunt those down soon too.

I'm using the uBoot from your Go-updater for all of these and it works without issue.

I also found we can set the kernel serial port to 115200, and then only u-boot needs to be set to match
find Yun in /target/linux/ar71xx/image/generic.mk  and set  `CONSOLE := ttyATH0,115200`

115200  will probably break the yun-bridge  as it expects the higher baudrate between the AVR and the AR9331

[821-serial-core-add-support-for-boot-console-with-arbitr.zip](https://github.com/arduino/lede-yun/files/2303017/821-serial-core-add-support-for-boot-console-with-arbitr.zip)

DanielRIOT, 2018-08-28:
I tested the ported patch on my Yun and it seems to work as expected ( baudrate at 250000 )

satmandu, 2018-10-22:
Could someone upload an 18.06.x image for the Yun? Or will there be an official one?

satmandu, 2019-01-26:
@DanielRIOT Any chance you'd be willing to put up a diff to the 18.06.x tree on github showing the changes you had to get a minimal working build for the yun?

The 18.06.1 release sysupgrade file is missing enough yun specific stuff (such as required kernel patches to get avrdude to work) to not connect easily to the arduino side.

DanielRIOT, 2019-01-28:
@satmandu will try and put up a repo later this week - You'll have to build it yourself to get the yun specific stuff in there ( fixed baudrate, AVRdude, Yun conf and Yun scripts) 
- I have not tried getting JUCi ( The "desktop" new Yuns present) or the Arduino web frontend ( classic arduino one ) up as I use the standard Openwrt frontend and I dont use the REST interface to control the AVR, but network programming works OK

satmandu, 2019-01-28:
Thanks. The Yun specific stuff in the tree aside from JUCi and the Arduino frontend area really all that I am looking for.

DanielRIOT, 2019-02-01:
@satmandu , 
* I cloned the master repo from openwrt ( https://github.com/openwrt/openwrt/tree/master ) and set up my build as [this ](https://oldwiki.archive.openwrt.org/doc/howto/build) instructs ) 
* Then added the arduino packages to feeds.conf.default ( https://github.com/arduino/openwrt-arduino-packages ) ( you can use [this ](https://github.com/arduino/lede-yun/blob/master/feeds.conf.default)as an example )
* called `./scripts/feeds update -a` and `./scripts/feeds/install -a` to get them into my "packages" directory and so they show up in menuconfig.
* added `yun-scripts` `yun-bridge` and `yun-conf` and `luci-app-arduino-webpanel`
* also needed to add the serial baudrate patch to the new kernel ( 4.14 ) in` /target/linx/ar71xx/patches-4.14`
[951-MIPS-serial-core-add-support-for-arb-baudrates.zip](https://github.com/arduino/lede-yun/files/2822700/951-MIPS-serial-core-add-support-for-arb-baudrates.zip)

* In menuconfig select the target system Atheros ar7xxx/ar9xxx  , Subtarget : Generic ,  target profile as Arduino Yun 

You might have to update u-boot ( [using the go-updater]( https://www.arduino.cc/en/Guide/YunUpdater ) - it works really well ), and then load your new image with ssh ( scp runs if you have ssh open, so you can use winSCP in windows or the regular linux scp to copy it to /tmp)

* to flash : run `sysupgrade -v -F /tmp/openwrt-ar71xx-generic-arduino-yun.bin` from terminal ( for some reason the magic number and "board name" doesn't get included or is calculated wrong and then the regular sysupgrade and web upgrade do not accept the new image and thus you need the Force `-F` flag

you might need to see compare which packages to include in your build to get the same behavior as Arduino's Yun build ( 
I haven't focused on recreating the Yun build completely - Yun is a convenient ar71xx platform for me to experiment on with OpenWrt specific stuff

all of this might brick your Yun during any stage (happened to me lots of times and its recoverable as long as you don't wipe out u-boot or ART partition on the flash, but you can always use something like [Flashrom ](https://www.flashrom.org/Flashrom), or a [arduino due/zero / any 3v3 device](https://github.com/LowPowerLab/SPIFlash) to read out or reprogram the flash on the Yun to put u-boot and ART back assuming you backed it up beforehand - you can do that from terminal - [check out this wiki ](https://openwrt.org/docs/guide-user/installation/generic.backup))

The "Sysupgrade" image that is built by the regular build can be uploaded to the flash from u-boot via tftp directly to the boot memory location ( 0x9f050000 ) - check the u-boot bootargs to get the correct location - it also has a lot of other things set up in there that give a useful insight into the system as it differs from the "regular" ar71xx targets

satmandu, 2019-02-01:
Thanks. That parallels much of what I was doing. It is reassuring to hear that you did much of the same, and yes I noted that annoyance with having to run `sysupgrade -v -F ` too.

Now that 18.06.2 is rolling out I'm tempted to just use https://github.com/openwrt/openwrt/tree/openwrt-18.06  as a base repo to work off of.

satmandu, 2019-02-01:
@DanielRIOT You mean `./scripts/feeds install -a` not `./scripts/feeds/install -a`  .

DanielRIOT, 2019-02-04:
Hi @satmandu , i did and then removed yunbridge from the build ( I don't use it at all - i use SPI to speak to the ATmega ),

 ( we might need to relook at the Arduino packages and sort out the dependencies and installations so these clashes don't happen )

DanielRIOT, 2019-02-04:
> Thanks. That parallels much of what I was doing. It is reassuring to hear that you did much of the same, and yes I noted that annoyance with having to run `sysupgrade -v -F ` too.
>
> Now that 18.06.2 [...]

I've also been looking into porting it o the ATH79 branch ( there are discussions on porting ar71xx platforms to ATH79 and devicetree so its more compatible with upstream linux developments  to rely on devicetree descriptions rather than a multitude of platform and board specific code snippets and files - to add or maintain a new board it takes about 4-5 file entries  in /target/linux/ar71xx, with ATH79 should only be the devicetree file )

satmandu, 2019-02-04:
> Hi @satmandu , i did and then removed yunbridge from the build ( I don't use it at all - i use SPI to speak to the ATmega ),

Ok that's interesting. Do you use anything besides the default avrdude configuration to write to the ATmega? Any suggestions on a good way to get data out of the ATmega without bridge using SPI?

(Any specific arduino libraries that you find useful for that?)

DanielRIOT, 2019-02-12:
I put the AVR into SPI slave mode and "command" it from Linux
remember to set GPIO 21 and 26 to the right state before sending data.

`/dev/spidev1.0 ` is created by the Yun boot in  [This](https://github.com/arduino/openwrt-arduino-packages/blob/for-lede-next/yun-scripts/files/etc/init.d/create-spi-gpio) script using the spi-gpio-custom command.

 ( from a ATH79 port I'm busy with ) :
ar9331 GPIO : Yun function

- 26	: bitbang SPI SS ( /dev/spidev1.0 ), but manually controlled via sysFS GPIO
- 27	: bitbang SPI MOSI  ( /dev/spidev1.0 )
- 8	: bitbang SPI MISO  ( /dev/spidev1.0 )
- 11	: bitbang SPI SCK  ( /dev/spidev1.0 )
- 21	: SPI level shifter enable
- 19	: AVR handsake signal ( has 10k pullup )
- 6	: GP6 on pcb
- 22	: handshake and GP6 level shifter enable
- 23	: UART level shifter enable
- 18 : AVR reset
- 20	: configuration button
- 1	: LED0 : Wireless LED (blue)
- 0	: LED1 : USB/System Led (White)
- 17	: LED6 : WAN led (yellow)
*/

The code was adapted form [AVRdude](https://github.com/arduino/openwrt-arduino-packages/blob/for-lede-next/yun-scripts/files/usr/bin/run-avrdude), [Reset MCU](https://github.com/arduino/openwrt-arduino-packages/blob/for-lede-next/yun-scripts/files/usr/bin/reset-mcu) and [gpio conf](https://github.com/arduino/openwrt-arduino-packages/blob/for-lede-next/yun-scripts/files/usr/bin/get-gpio-conf)

gpio 21 should already be an output  GPIO in /sys/class/gpio... 
gpio export 26
gpio direction 26 out
gpio output 26 1

# our transaction
gpio_output 21 1 // 21 HIGH :  enable level shifter
gpio_output 26 0 // 26 LOW  : CS low to speak to the AVR
# echo -n -e '\xC0\x0F\x0F' > /dev/spidev1.0 // send some bytes
myCapp > /dev/spidev1.0 // send some bytes from a C application's output
gpio_output 26 1 // 26 HIGH  : CS high to deselect  AVR
gpio_output 21 0 // 21 LOW :  disable level shifter

noblepepper, 2019-02-20:
Hi! Thanks for the baud rate patch. Glad to find some others working on the OpenWrt side of the yun. I also backported the baud patch to 18.06 (4.9 kernel) on my repo https://github.com/noblepepper/openwrt-trunk/tree/yun-18.06. 

I have pepe2k's u-boot_mod running on my v1 yun. His u-boot image for the carambola2 booted out of the box I just configured for the wlan button and leds. With this you get web image load and netconsole activated by the wlan button and CHANGEABLE baudrate. If you want to use stock openwrt rates you can set uboot to 115200 or 230400. I haven't molested by v2 yun with this yet.

vwone, 2019-05-27:
Hi,
I am originally a user of the Yun rev1 with linino, I used the serial link /dev/ttyATH0 to quickly communicate between the processor and the microcontroller at 250 kbps.
I also noticed that my code no longer works with LEDE and Yun rev2 (while yet it works with rev2 and linino) :
[https://forum.arduino.cc/index.php?topic=557667.msg4174964#msg4174964](url)

so I published a code that works under Linino but not Lede:
[https://github.com/vwone/Yun-Atheros-ATmega-UART-communication](url)

So I'm looking for a solution to my problem and if I understand correctly, there are two:

1. The first is to apply a patch. But it seems to me very complex and will surely ask me a lot of time because I am novice.

2. The second, which seems a little simpler, is to use the SPI.

According to your post above @DanielRIOT:
[https://github.com/arduino/lede-yun/issues/4#issuecomment-462707702](url)

I wrote an exchange program between the processor and the microcontroller:

[https://github.com/vwone/Yun-Atheros-ATmega-SPI-communication](url)

Sadly :

- Unlike my UART code, it works with Lede but not with Linino,

- it does not handle the frames because the transmissions are byte by byte, 

- and it does not reach the requested speed (0.6 MHz instead of 4 MHz).

Could you please @DanielRIOT share your SPI code?

In parallel, I will try to follow these instructions to apply the patch with Ubuntu or Debian stretch :
[https://github.com/arduino/lede-yun/issues/4#issuecomment-459816752](url)

Thank you for your feedback.

DanielRIOT, 2019-06-04:
Hi @vwone , 
Applying the patch would require you to rebuild openwrt, its fairly simple once you get the toolchain set up but can be daunting the first time round. [Setting up toolchain](https://openwrt.org/docs/guide-developer/build-system/install-buildsystem), [using patches](https://openwrt.org/docs/guide-developer/build-system/use-patches-with-buildsystem) 

regarding the SPI communication :
* its created by [this file](https://github.com/arduino/openwrt-arduino-packages/blob/for-lede-next/yun-scripts/files/etc/init.d/create-spi-gpio)  when openwrt boots, you could create a similar one in init.d and see what happens :)
* its bit banged (software) and Linux seems to limit the max to around 800kHz even if you set it higher.

I don't know if the Linino branch has this file, but in the Arduino Yun branch it isn't defined for Linino so if you load it it wouldn't create the port.
You can run it  manually, or add it in your system in one of the startup files in etc/init.d.

* insmod spi-gpio-custom bus0=1,11,27,8,0,125000 : would have SPI1 with  MISO=8, MOSI=27, SCK=11, CS is manually handles via sysfs - GPIO 26 on Yun, max SPI freq = 125 kHz)  [source here](https://github.com/openwrt/openwrt/tree/master/package/kernel/spi-gpio-custom/src)
* I think this would automatically give you the /dev/spidev1.0 ( you might need  kmod-spi-dev module too )

I usually prefer creating my own frames : [STX], [frame size], [Payload 0- 255], [ETX], [Checksum] - a habit form working with Rf modems or Xbee in API mode, even if i only send 10 bytes. if the frame size is one byte the payload can be up to 255 bytes and i can allocate a static buffer in Arduino for 259 bytes so there can never be an overflow.

 Unfortunately I have not found a bitbanged UART for the AR9331, and it only has 1 UART which I use for Console and I have not tried adding another CS to the hardware SPI ( which is used for SPI flash access - it runs at 25MHz :D ), so we're stuck with bitbanged SPI, bitbanged I2C or the console UART with Bridge - or disabling console output and using the regular UART- like in your arduino forum post )

satmandu, 2019-08-16:
@DanielRIOT Did you ever get any further with your ATH79 port, since ar71xx is now source only on OpenWRT?

DanielRIOT, 2019-08-19:
@satmandu , I have a working Devicetree that brings the Yun in as a ATH79 device - I hav etnot tested any of the Arduino specific applications ( the "desktop" web interface and yun-bridge ) as I dont use those at all - Yun is a readily accessible AR9331 platform for me.

I'll create an openwrt branch on my other ( non work ) repo and link to it from here that only has Yun changes ( I have about 5 different devices that have very specific hacks / fixes which won't be accepted in openwrt ).
 Then we can submit a pull request to openwrt for integration for ATH79 Yun

satmandu, 2019-09-20:
Have you had any time to get that ATH79 yun branch up?

DanielRIOT, 2020-01-14:
hi @satmandu 
There is currently a PR on teh main Openwrt repo to port Yun to ATh79

https://forum.openwrt.org/t/port-arduino-yun-to-ath79/52555

satmandu, 2020-01-14:
Thanks for the update. I've retired my Yun at this point as the flash memory on mine seemed to be getting to be unreliable, but I'm glad that people are keeping this hardware alive.

mans0n, 2020-01-19:
For anyone interested, I've made a PR for ath79 target on OpenWrt mainline. openwrt/openwrt#2692

WhidbeyBill, 2020-03-02:
Hello,
I am a casual user of all three versions of the YUN. The Rev2 has a peculiar arduino:blue:led LED malfunction. The arduino:blue:led is preprogrammed - somewhere - to remain on steady when the WLAN is active, connected or not. I have tried multiple times, In LUCI, to set the blue LED function to netdev. wlan0, trigger "tx rx" or just "phy0rx" and observe the code changes in the /etc/config/system file.  However after about 30 seconds it reverts to the original steady-on condition. The system file does not change.
 afaik, this has been a default since it was new two years ago. I can add the white function, as shown, and the white LED continues to function as configured. 
Thank You for your interest. The OpenWrt forum seems more active than the YUN on arduino.cc. 
[system.txt](https://github.com/arduino/lede-yun/files/4277082/system.txt)

nsmith1024, 2020-12-28:
Im trying to build OpenWRT for YUN rev-2 board. When i type make menuconfig the sub-menu shows "Arduino YUN" in the settings, is that for rev-1 board or rev-2? because it doesnt say which

nsmith1024, 2021-01-23:
Wow, no response, awesome!! why is there so many different versions of OpenWRT YUN, how are we supposed to know which one to use? This repo hasnt been changed since 2018, is this the correct repo, or is it http://github.com/openwrt/openwrt which also has ARDUINO YUN in its menu?

DanielRIOT, 2021-01-25:
I'm not sure what the electrical / connection difference is between Rav2 and the original one, but there are differences between this repo and the Openwrt "base" one, mostly :
 - Serial Baud rate is 115200 in OpenWRT, but 250000 in the Arduino port

- Arduino port has the Bridge library included so one can speak with the  arduino bridge library that can be used on the 32u4 ( in OpenWRT you need to use SPI or the UART and make up your own protocol between the Micro and Linux - SPI is easier, for me at least,  as the UART is also used for the Linux Console and is very busy ) 
- Arduino has AVRDude preinstalled to flash the 32u4 from linux on the AR9331
- Arduino's OPenWRT fork has a "web Desktop" installed in addtition to teh base OPenWRT LuCI system
- There has been some work done in porting most of the functionality into mainstream OpenWRT, I have not followed if that has been incorporated into ATH79 branch yet - search the openWRT forum for " Yun" and you can get the discussion

You can flash a Yun with base ( probably recompiled using their build system from ar71xx or ATH79 if its been ported already ) OPenWRT but will have to play around with some of the firmware on the 32u4 to get to the linux console ( set it to 9600 baud if 115200 or 250000 don't work as the driver in linux falls back to that if the uboot bootargs tell it to use an unsupported rate ) - look at the arduino code for the yun-serial example, you will have to add 9600 support..