arduino/lede-yun #6: Is this for Arduino YUN Rev-2??

nsmith1024, 2020-12-26:
Hello,

Im trying to find the official OpenWRT for Arduino YUN rev-2 hardware?

Is this the one, or can you tell me the git hub link to it?

Thanks

facchinm, 2020-12-28:
It's this one; if you just want the feed packages for recompiling vanilla openwrt you can add https://github.com/arduino/openwrt-arduino-packages to your `feeds.conf`

closed 2020-12-28 by facchinm

nsmith1024, 2020-12-28:
Do you know where is the repository for the OpenWRT Arduino YUN rev-2 board? Not the packages, i mean the linux OS code itself for the rev-2 board?

facchinm, 2020-12-28:
This exact repository is the one you are looking for. Please stop spamming all the other repos

nsmith1024, 2020-12-28:
Sorry i was not trying to spam, i ask so many forums and nobody answered, i even asked on the arduino forum itself, nobody answered, so i started asking around.

nsmith1024, 2021-01-03:
> 
> 
> This exact repository is the one you are looking for. Please stop spamming all the other repos

I tried building openWRT for YUN using this repo but it has a lot of missing dependencies

1. What is the difference between this repo and http://github.com/openwrt/openwrt?

I tried using http://github.com/openwrt/openwrt, it also has Arduino YUN already in the settings if you select target=ARTH79 so why do we need two repos for OpenWRT YUN?

2. However when i build that version of OpenWRT the build size is only 5.4M, isnt it supposed to be about 16M?

Any ideas why their build is only 5.4M?

nsmith1024, 2021-01-23:
I ask question and get no reply, then if i ask in another forum you say im SPAMMING? so how long should i want for an answer before i ask somewhere else?

OtterCodes101, 2024-11-30:
dude just use this one